dl1bz/deskhpsdr #1: Compile Error on MacOS 15.2

SDupuie, 2025-01-19:
Received the following error while attempting to compile on Mac OS 15.2.

cc -O3 -Wall -Wextra -Wimplicit-fallthrough -Wno-unused-parameter -Wno-deprecated-declarations -pthread -I./src -D MIDI       -D__LDESK__     -DPORTAUDIO  -DTCI -D GIT_DATE='"2025-01-19"' -D GIT_VERSION='"v2.5.4"' -D GIT_COMMIT='"df1be93"' `pkg-config --cflags gtk+-3.0 --cflags json-c` -I./wdsp `pkg-config --cflags portaudio-2.0`  -c -o src/tci.o src/tci.c
src/tci.c:37:10: fatal error: 'openssl/sha.h' file not found
   37 | #include <openssl/sha.h>

Openssl appears to be installed in the Homebrew path (/opt/homebrew/) as expected. The include directory has a subdirectory "openssl" and in that directory there is a file called "sha.h". So, I think the file is where it is supposed to be.

I got around this issue temporarily by disabling TCI support. The program then compiled and I was able to start it.

dl1bz, 2025-01-19:
Have no problems here if compiling. Do you installed this: `brew install openssl@3` ?
Thats a requirement for using TCI. Maybe try `brew update && brew upgrade` for bring Homebrew most up-to-date.

73 Heiko, DL1BZ

SDupuie, 2025-01-19:
I tried brew update && brew upgrade before I posted the issue. For completeness, I just tried brew install openssl@3 but it said the package was already installed, so I did a reinstall. Still have the same problem. It's probably a path issue.

dl1bz, 2025-01-19:
Difficult to say, why the compiler don't find this file. That's normally a problem at OS level or better around Apples CLang Compiler and Linker and/or it's environment - the program code cannot be wrong, otherwise I had the same problems like you.
But I havn't.

You can try this:
In the `tci.c` change 
```
#include <openssl/sha.h>
#include <openssl/evp.h>
```
and give the full path to the `sha.h`, e.g.
`#include "/opt/homebrew/Cellar/openssl@3/3.4.0/include/openssl/sha.h"`
For the `evp.h` I think you need to do the same. Try too with the full path.
Let me know, if it helps. At the moment I can check only macOS 14.7.2. Here it works. Tomorrow I can check macOS 15 too. I check Linux too (PiOS 64) with my Raspi5, there are no compiling problems too.

SDupuie, 2025-01-19:
Well, this is what I did to solve the problem. I edited the Makefile to replace:

INCLUDES=$(GTKINCLUDE) $(WDSP_INCLUDE) $(AUDIO_INCLUDE) $(STEMLAB_INCLUDE)

with

INCLUDES=$(GTKINCLUDE) $(WDSP_INCLUDE) $(AUDIO_INCLUDE) $(STEMLAB_INCLUDE) $(TCI_INCLUDE)

All is well now.

73,
Scott
AK5SD

dl1bz, 2025-01-20:
Ok, you are right, your patch was correct. The TCI extension was a merged patch came from piHPSDR, DL1YCF starts with TCI in piHPSDR and I take this over later in deskHPSDR, because DL1YCF and I work together a longer time at the TCI implementation. Later I extend TCI, but only in deskHPSDR. Up to now it is a work in progress, I'm not ready with it, but it is usable from now. But Audiostreaming (starts with TCI 2.0) we will not implement, not now and not later. Only things called "classical" CAT for control the TRX or better the SDR application.

I talked with DL1YCF and he confirm, his Makefile was incomplete. He also patched his Makefile from piHPSDR too and I patched the same way like piHPSDR now. So think all is correct now in both versions, piHPSDR and deskHPSDR.

But anywhere, normally the compiler must found all things in /usr/local/include by itself and you don't need to do an extra configuration if compiling. Can be a problem with environment variables like CPATH and other. You should check your shell environment if there something is wrong or has wrong settings.

SDupuie, 2025-01-20:
I renamed my custom Makefile, did a `git pull` followed by `make clean` and `make app`. It compiled without error. Problem solved.

73,
Scott
AK5SD

dl1bz, 2025-01-20:
Please do not edit the Makefile itself, use the file make.config.deskhpsdr instead and put the options in it you need.
To prevent this file against git pull, do the following:
`$ git update-index --assume-unchanged make.config.deskhpsdr`

After this the file is excluded from git and you can make changes in it.

closed 2025-01-20 by dl1bz