dl1bz/deskhpsdr #4: Make Error

W1IZZ, 2025-03-21:
Updating to latest version, get an error: "src/sliders.c:719:47: error: ‘autogain_en’ undeclared (first use in this function); did you mean ‘auto_tune_end’?"

Also: "src/sliders.c:719:61: error: ‘autogain_en_signal_id’ undeclared (first use in this function)
  719 |     g_signal_handler_block(GTK_TOGGLE_BUTTON (autogain_en), autogain_en_signal_id);"

Make: *** [Makefile:488: src/sliders.o] Error 1

Undeclared variable?  I first tried the update process, which failed.  I then loaded a clean copy and tried again--same error.

dl1bz, 2025-03-21:
I fixed the issue. Please do a git pull and make clean. The compile again.

73 Heiko, DL1BZ

W1IZZ, 2025-03-22:
Thanks!  Works perfectly and I like the new features.

dl1bz, 2025-03-22:
I had done some code cleanup yesterday and forgot to exclude a code part if set `AUTOGAIN=OFF` (which is the default setting) in the `make.config.deskhpsdr`. I'm using `AUTOGAIN=ON`, so this issue wasn't present in my case. Thanks for your notice.
I'm check and correct some things now and again, so please do again a git pull for latest update and recompile. Let me know, if there no more issues now.
Notice, because it is a problem with git, which I cannot compensate: If git pull fails, because there is a problem with the `make.config.deskhpsdr`, so delete this file and repeat git pull. After this, you need re-adjust the  `make.config.deskhpsdr` with your needed settings and recompile all. Don't forget first a `make clean` before `make`, sometimes old .o files will be not updated and then the linker has a problem. A `make clean` delete all *.o files and `make` generate all new completly.

73 Heiko, DL1BZ

closed 2025-03-23 by dl1bz