dl1bz/deskhpsdr #5: make error

DL1EGL, 2025-03-22:
I have just downloaded the latest version of deskhpsdr and get the following error message after entering make:

In file included from src/actions.c:28:
src/sliders.h:37:12: error: missing binary operator before token "("
   37 | #if define (__AUTOG__)
      |            ^
make: *** [Makefile:492: src/actions.o] Error 1

VY 73, Robert

dl1bz, 2025-03-22:
I was update the master branch yet. Please do a make clean, git pull and recompile. If git pull have a problem with the make.config.deskhpsdr, please delete this file and repeat git pull. After this please adjust the make.config.deskhpsdr again with your settings you want before compiling new.
Let me now, if works all now.

73 Heiko

DL1EGL, 2025-03-22:
Problem solved. Thank  Youtube for the support. VY73

closed 2025-03-22 by DL1EGL