dl1bz/deskhpsdr #6: Failing to compile on Linux

bluzee, 2025-03-23:
Getting this error....

In file included from src/new_menu.c:70:
src/screen_menu.h:24:8: error: unknown type name ‘bool’
   24 | extern bool is_valid_rgb(const char *str);
      |        ^~~~
src/screen_menu.h:25:8: error: unknown type name ‘bool’
   25 | extern bool is_valid_hex(const char *str);
      |        ^~~~
make: *** [Makefile:492: src/new_menu.o] Error 1
make: *** Waiting for unfinished jobs....

I'm not exactly certain what the issue is or if this is the correct fix but this patch sorts it and gets it to compile again...

diff --git a/src/screen_menu.h b/src/screen_menu.h
index b690b5b..590205e 100644
--- a/src/screen_menu.h
+++ b/src/screen_menu.h
@@ -21,5 +21,5 @@
 
 extern void screen_menu(GtkWidget *parent);
 extern void schedule_apply(void);
-extern bool is_valid_rgb(const char *str);
-extern bool is_valid_hex(const char *str);
\ No newline at end of file
+extern _Bool is_valid_rgb(const char *str);
+extern _Bool is_valid_hex(const char *str);

The background hex still appears to work on the screen menu and entering an incorrect rgb hex entry just returns you to the previously set rgb hex. I assume this is intended behaviour.

dl1bz, 2025-03-23:
That's a compiler problem.
If you use bool you need to include `<stdbool.h>`, which IS included in the screen_menu.c. The same with the used function isxdigit() inside the function is_valid_hex(), here we need include `<ctype.h>`. 
Some newer compiler (C99 standard) have use instead bool now `_Bool` and `<stdbool.h> `is no more available, as far as know now as build-in, which means, this doesn't require no more an external include-file for using this. I need to support a wide range of compilers, so I think, the better choice will be to switch from bool to gboolean, that's a part of the GTK environment and is equal to bool. I will change this in the evening, now I'm out and will be back later the day.

So it is not a real issue - all depends from the used compiler. The LLVM-compiler I use here with macOS works correct with my code and my Pi5 with PiOS64 (gcc) hasn't too a compiling problem with this code.

73 Heiko, DL1BZ

bluzee, 2025-03-24:
Yes. gboolean works. Closing issue as fixed.

closed 2025-03-24 by bluzee