dl1bz/deskhpsdr #10: Maybe don't uninstall the program with make clean

bluzee, 2025-04-13:
Well, at least this confused me for a while.  I always compile, cp the new binary to /usr/local/bin and then run a make clean.   I think moving the program uninstall  to a make uninstall would be better understood.

dl1bz, 2025-04-14:
There was a reason, why I changed this in Makefile. I got more than one feedback from users, they do a git pull, recompile all  - but forgot to replace the old binary with the new compiled. Then I got the notice from this users "deskhpsdr was not updated". Thats why I changed in the Makefile the step with remove too the binary from `/usr/local/bin` (if Linux) & remove deskhpsdr.app from the desktop (if macOS) if they had done a `make clean`.
The more used popular option in Makefiles is `make distclean` instead of `make uninstall`. That is equal to your request and the better solution. I can change this, no problem, but I think, my solution was the easier way for users with not so much and deep knowledge.

So if you think, we should have both options, `make clean` and `make distclean`, confirm this here and I will change it again.
Up to this I let this "issue" open.

73 Heiko, DL1BZ

bluzee, 2025-04-14:
Alright, so the step they are forgetting is actually **make install** which you already have.  

make clean  should remove compiled objects only from src directory.
make distclean is to additionally remove config files from src directory ie. the product of ./configure script.  Not really applicable to deskHPSDR.
Neither should remove any file from the main filesystem tree. 
make uninstall is also not really necessary as so few files are actually installed for deskHPSDR.

If you feel it helpful to delete files for Mac users in make clean then so be it. In that case my suggestion would be to remove the Linux part of that section from make clean.  make install is all that is needed.  Having the program deleted from the system by make clean is not expected behaviour.  sudo permissions are generally held for a short time so if run shortly after using sudo to install then unfortunately the delete also succeeds if you run make clean in that time frame.

dl1bz, 2025-04-14:
[https://www.gnu.org/software/make/manual/html_node/Standard-Targets.html#Standard-Targets](https://www.gnu.org/software/make/manual/html_node/Standard-Targets.html#Standard-Targets) give the right answers. So we need `make uninstall` - that is the correct choice as you recommended, I agree with you. I will change the Makefile in this direction.

73 Heiko, DL1BZ

bluzee, 2025-04-14:
OK, very good.  Closing issue.

closed 2025-04-14 by bluzee