dl1bz/deskhpsdr #23: TX Bandwidth does not follow filter setting when TX EQ is enabled

joshuajayg, 2025-04-24:
Version 2.6.28, built from source on 2020 Macbook Pro Intel i5.   Bug exists in 2.5.8b build as well.
Hermes Lite 2 with N2ADR board.

**Bug**
When the transmit equalizer is enabled, transmit bandwidth filter selections are completely ignored.  
If the maximum frequency in the transmit equalizer is 3500Hz, transmit bandwidth cutoff begins at 3500Hz regardless of the filter selection being 2100Hz or 4400Hz in the TX Menu.

TX EQ should be independent of filter selection to allow quickly changing bandwidth without changing microphone EQ profiles.

If the transmit equalizer is disabled, TX Menu filter settings are followed.

joshuajayg, 2025-04-24:
I booted my Macbook in Windows and checked the Hermes Lite build of Thetis.  The transmit equalizer was active and the top highest frequency was 16kHz.  Transmit bandwidth is adjustable completely separately from the EQ frequencies.

I also started my Raspberry Pi5 running piHPSDR version 2.5 using WDSP version 1.25.  With transmit EQ enabled and the highest frequency set to 5000Hz, transmit bandwidth is still controlled independently using the TX Menu or from tying it to the RX bandwidth and changing the RX bandwidth.  

So far deskHPSDR is the only program of the three I just tried that relies on the TX EQ to be set to your desired bandwidth when using the EQ.

dl1bz, 2025-04-25:
I'm not sure, if this really is a bug or issue. The whole signal processing will be done inside of the WDSP. The settings we do (in menu, slider and so on) is set only parameter to functions of this WDSP. But there are depencies between some functions inside the WDSP.

In the wdsp subdirectory of my repo you will found the WDSP_Guide_V1.25.pdf, which is the manual for developers, how to use this WDSP library. First have a look at page 69, you will see the whole block diagram of the TX chain for understand, how the single parts of this chain are connected, placed and how the signal is processed in the case of TX.

For my understand I tend to agree with @Richz28 , if we set with the TX EQ an upper limit then the bandpass (=TX BW filter) cannot be wider open, because this part is _behind_ the TX EQ and affects the signal only after the TX EQ and not before or instead.
But we have 3 Bandpasses, not only one. Look at page 89 of the WDSP manual for a description of the 3 bandpass filter. **Have an attention to the notice for filter 1 and 2: "The ON/OFF state is automatically set inside WDSP".**

So it can be possible, that a race condition change the filter behavior (CFC, CESSB etc.). For a deeper investigation I need to have a look in your logfile, which you can found in your working-directory $HOME/Library/Application Support/deskHPSDR (if macOS) and called `deskhpsdr.stdout`. There is a lot of debug output. A look in your device .props file can also be helpful for me.

The difference between pihpsdr and my deskhpsdr is, I made some options accessible and adjustable for the user, which are only hardcoded in pihpsdr. But I will compare both versions in this direction. Can you provide me with the files mentioned?

73 Heiko, DL1BZ

dl1bz, 2025-04-25:
Additional notice; I setup the TX EQ up to 4kHz and if I change the TX filters upper limit from 4000Hz down to 2400Hz I see this effect  in the TX spectrum (live transmission and reported with my SDRplay as external RX via RF). There is a cutoff came from the TX filter and will be changed with the TX filter. So as a first discovery I cannot say, that the TX filter is bypassed if the TX EQ is ON. The filter work and has an influence inside the chain.

73 Heiko, DL1BZ

joshuajayg, 2025-04-25:
Heiko, thank you for pointing out the manual. I have attached the log file.  Below that is `audio_profile_1.prop`  I transmitted with a few different filters using white noise so you can see the filter changes in the log.

I found the same result as you, if the filter selection is narrower than the eq high limit, the bandwidth is narrowed.  
Is it possible to have a wider EQ bandwidth but limit the transmitted bandwidth in the TX menu or through tying it to the RX filter.  I did not thing that was the case yesterday, my misunderstanding.  I was testing the opposite condition.

This is an acceptable condition in my opinion.  It does look like there is an option to not make the EQ limiting.  Page 80 of the WDSP manual gives a call for roll-off outside the specified EQ band.  `void SetTXAEQCtfmode (int channel, int mode)`. 

deskhpsdr.stdout:
```00:00:00.008 startup: working dir changed to /Users/joshuagriffin/Library/Application Support/deskHPSDR
00:00:00.008 main: gtk_application_new: org.dl1bz.deskhpsdr.pid1412
00:00:00.243 Build: 2.6.28 (Branch: master, Commit: b82525c, Date: 2025-04-23)
00:00:00.243 GTK+ version 3.24.43
00:00:00.243 sysname: Darwin
00:00:00.243 nodename: Joshuas-MacBook-Pro.local
00:00:00.243 release: 24.3.0
00:00:00.243 version: Darwin Kernel Version 24.3.0: Thu Jan  2 20:22:00 PST 2025; root:xnu-11215.81.4~3/RELEASE_X86_64
00:00:00.243 machine: x86_64
00:00:00.244 load_css: failed to load CSS data from file default.css
00:00:00.245 load_css: hard-wired CSS data successfully loaded
00:00:00.245 create top level window
00:00:00.250 Monitor Number within Screen=0
00:00:00.250 Monitor: width=1440 height=900
00:00:00.250 display_width=1280 display_height=600
00:00:00.254 create grid
00:00:00.255 add grid
00:00:00.255 add image to grid
00:00:00.255 create pi label
00:00:00.255 create build label
00:00:00.255 add build label to grid
00:00:00.255 create status
00:00:00.255 add status to grid
00:00:00.442 g_idle_add: init
00:00:00.571 init
00:00:00.572 audio_get_cards: PORTAUDIO call audio_get_cards
00:00:00.716 audio_get_cards: OUTPUT DEVICE, No=0, Name=MX259
00:00:00.726 audio_get_cards: OUTPUT DEVICE, No=1, Name=USB AUDIO  CODEC
00:00:00.775 audio_get_cards: INPUT DEVICE, No=2, Name=USB AUDIO  CODEC
00:00:00.828 audio_get_cards: INPUT DEVICE, No=3, Name=iPhone Microphone
00:00:00.837 audio_get_cards: OUTPUT DEVICE, No=4, Name=External Headphones
00:00:00.888 audio_get_cards: INPUT DEVICE, No=5, Name=MacBook Pro Microphone
00:00:00.901 audio_get_cards: OUTPUT DEVICE, No=6, Name=MacBook Pro Speakers
00:00:00.981 audio_get_cards: INPUT DEVICE, No=7, Name=VB-Cable
00:00:01.006 audio_get_cards: OUTPUT DEVICE, No=7, Name=VB-Cable
00:00:01.006 Securing wisdom file in directory: /Users/joshuagriffin/Library/Application Support/deskHPSDR/
00:00:01.439 loadProperties: protocols.props
00:00:01.546 old_discovery
00:00:01.547 discover: looking for HPSDR devices on en0
00:00:01.547 discover: bound to en0
00:00:01.547 discover_receive_thread
00:00:03.548 discovery: bytes read -1
00:00:03.549 old_discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
00:00:03.549 discovery: exiting discover_receive_thread
00:00:03.549 discover: exiting discover for en0
00:00:03.550 discover: looking for HPSDR device with IP 192.168.42.23
00:00:03.551 discover_receive_thread
00:00:05.552 discovery: bytes read -1
00:00:05.552 old_discovery: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
00:00:05.552 discovery: exiting discover_receive_thread
00:00:05.552 discover: exiting HPSDR discover for IP 192.168.42.23
00:00:05.552 Trying to detect via TCP with IP 192.168.42.23
00:00:08.552 discover: select() timed out on TCP discovery socket
00:00:08.552 discovery found 0 devices
00:00:08.552 old_discovery: macOS = 1 Raspberry Pi = 0 Lokal = 0
00:00:08.683 new_discover: looking for HPSDR devices on en0
00:00:08.683 new_discover: bound to en0 192.168.42.19 255.255.255.0
00:00:10.685 new_discover: bytes read -1
00:00:10.685 new_discover: recvfrom socket failed for discover_receive_thread: Resource temporarily unavailable
00:00:10.685 new_discover: exiting new_discover_receive_thread
00:00:10.685 new_discover: exiting discover for en0
00:00:10.692 discover: looking for HPSDR device with IP 192.168.42.23
[... 286 lines cut ...]
modeset.0.txeq.2=-8.000000
modeset.0.cfc_post.1=0.000000
modeset.0.cfc_lvl.1=0.000000
modeset.0.cfc_frq.1=50.000000
modeset.0.rxeqfrq.1=50.000000
modeset.0.rxeq.1=-3.000000
modeset.0.txeqfrq.1=30.000000
modeset.0.txeq.1=-9.000000
modeset.0.cfc_post.0=-9.000000
modeset.0.cfc_lvl.0=3.000000
modeset.0.cfc_frq.0=0.000000
modeset.0.rxeqfrq.0=0.000000
modeset.0.rxeq.0=0.000000
modeset.0.txeqfrq.0=0.000000
modeset.0.txeq.0=0.000000
modeset.0.cfc_eq=0
modeset.0.cfc=0
modeset.0.phrot_enable=0
modeset.0.lev_gain=0.000000
modeset.0.lev_enable=0
modeset.0.compressor_level=0.000000
modeset.0.compressor=0
modeset.0.en_rxeq=1
modeset.0.en_txeq=1
```

dl1bz, 2025-04-25:
Thanks for your feedback, it confirms my assumption for a correct function, which means no bug or issue. Thats why I add the WDSP manual for more clearness, how the used WDSP library is working. Especially the block diagrams are important for a fundamental understanding of all relationships in signal processing of the WDSP. In other words, all apps based on the WDSP should produce the same and identical results. The GUI of these apps is more or less only the console for the WDSP operation.

I think about to add the `SetTXAEQCtfmode()` function in the TX Menu and make this switchable ON or OFF. That is not so much work. So the user can decide by himself, if he want use it or not.

There are some other important things the user need to know, e.g. CESSB Overshoot Control. I recommed to set this ON (in deskHPSDR this is calling Auto CESSB in the TX Menu), but: "**ensure that the Speech Processor is turned ON as a prerequisite for this block**". My CESSB automatic do exact this, if the Speech processor (aka Baseband Compressor) is on, CESSB is activated. If the Speech processor is off, CESSB will be deactivated by the automatic. If a Digimode is selected like DIGL or DIGU, all audio parts will be switched off, because we need a linear AF frequency response for this, no EQ, no processor or similar things. In the case of using the TUNE function it is the same procedure.

In pihpsdr some of this parts of the audio TX chain are hardcoded and cannot be changed by the user. That was one of the reason, why I was starting with deskHPSDR after a very long, but unsuccessful discussion with Chris, the maintainer of pihpsdr. He is a "CW man", but does not fully understand the concerns of the SSB users like me. So I made in deskHPSDR the most functions accessible and adjustable for the user.

73 Heiko, DL1BZ

dl1bz, 2025-04-25:
Ok, you have the option TX-EQ Ctfmode in the TX Menu > TX Audio Tools.
Play around with this and give me feedback, if it works as we suspect.
Do a `git pull` first for update, it's uploaded in the master branch of deskHPSDR.
It's too late in Germany now for testing, hihi...

73 Heiko, DL1BZ

joshuajayg, 2025-04-25:
Just tested.  It sure does.  Thank you for listening to my inaccurate complaints.  Haha.  

I transmitted white noise with the filter set at 5kHz and the EQ maxing out at 3kHz.  With **TX-EQ Cftmode** set, the bandwidth was 5kHz.  When unset, the bandwidth decreased to 3kHz plus just a little.  

Joshua
KJ7LVZ

closed 2025-04-25 by joshuajayg

Richz28, 2025-04-26:
TX EQ is very effective keeping frequencies held within your TX bandwidth 99% of the time, if you're not over driving your amplifier into compression, meaning keeping linearity, others will not observe your signal on the panafall, blowing past the edges of your TX bandwidth filter, and that's without using pure Signal, with pure Signal, depending which amplifier and design you can improve even more when stations are nearby or very strong, there able to view your signal down -70db, and your still clean, in general would like to be able to save even more profiles , because in the USA we use AM wide AM Narrow, from 8k out to 16k, ssb 4k is the normal for most operators and we experiment voo doo, to full 8k ESSB, so I would mind more TX presets,, anyway, back to the point ,we do not want any changes to this TX EQ,. The CFC don't won't this way , it will only compress, frequencies selected within its bell , but it doesn't restrict TX Bandwidth, hope this is helpful 73