dl1bz/deskhpsdr #24: Pure Signal Auto Tune

GregVK3kv, 2025-04-25:
Hi Heiko,

I have found using Pure Signal Auto Antenuate, PS Map , Mon using Internal PS Feedback Antenna in VK3AMP when two tone to calibrate can struggle to continue to keep a clean two tone signals on display.

I have found if I undertake the auto Antenuate cycle after Restart can achieve a nice two tone then need to stop cycle unclick auto Antenuate to hold it. If I keep auto Antenuate ticked I can see my PS signal not achieving or holding sharp bandpass.

I never experienced this issue over 2 years on Thetis with my station a. u doing something wrong or its an issue I have?

I thank you for all your great work.

Cheers GregVK3KV

dl1bz, 2025-04-25:
All PS functions I leave untouched up to now from my pihpsdr fork in October 2024. Do you have the possibility to make a comparison with pihpsdr from DL1YCF ? The question is, do pihpsdr the same you wrote or not. You can compile pihpsdr without any additional work, the requirements are the same like with deskhpsdr. You also can run pihpsdr and deskhpsdr at the same machine, but not at the same time. deskHPSDR has an own working-directory, which is different from pihpsdr. So you can play with both apps installed, they don't bother each other.

I was using PS too, but I use RX3 (which is not the internal feedback, this need an extra antenna connector) as input.
For more information about RX3, look here [https://dl1ycf.darc.de/hl2.htm](https://dl1ycf.darc.de/hl2.htm), section "PURESIGNAL with external feedback". But Chris wrote a lot of interesting things around PS at his website.

After my 600W PA I have an RF sampler with -50db output, which is connected to RX3 for feedback. I can use the Auto ATT without any problems, I see no "struggling" in the feedback signal in my case.

Can you make a comparison between deskhpsdr and pihpsdr ? That's easier for me to investigate this issue, because I can compare my code with the code from pihpsdr and look for differences. I know, that Chris/DL1YCF spent a lot of time for optimization the PS function in pihpsdr. So let's see if we have a real issue or not.

73 Heiko, DL1BZ

GregVK3kv, 2025-04-25:
Thank you Heiko,

I downloaded the DL1YCF information and I believe I may have had one configuration setting incorrect for the ER Sampler signal which I will test tomorrow.

Thank you again for your assistance and I will let you know my findings.

Cheers GregVK3KV

dl1bz, 2025-04-25:
Ok, I let this issue as open for now and wait for your feedback.

73 Heiko, DL1BZ

GregVK3kv, 2025-04-26:
Hi Heiko,

I have followed PS instructions and suspect the set up for Auto Antenuate
may have an issue especially for SSB.

Without PS activated a two tone signals is pure 2 tones very clean showing
on TX scope display.

When PS auto Tuner activated with feed back ant ext 1, Min on, PS Map when
Two Tone activated many signal peaks across screen auto Antenuate TX Ant
shows 19.

If I untick auto Antenuate, adjust TX ATT to 31 transmit it shows
correcting and cleans up the bandpass.

The TX scope is slightly LL ess in height on the display however output
power is the same showing from amp.

It's interesting why auto Antenuate struggles to provide a clean sharp
skirt bandpass and why would TX scope on display reduce slightly in height
on the display?

It really does not make sense to me hope you may look into this as I
appreciate you haven't at this stage. Cheers GregVK3KV
Cheers,

Greg VK3KV

dl1bz, 2025-04-26:
Again my question, can you verify deskhpsdr against pihpsdr (DL1YCFs version), if the issue is equal in pihpsdr ? I need to know, if your issue exists in both versions (as I said, the PS code in deskhpsdr is untouched and was taken from pihpsdr in October 2024).

73 Heiko, DL1BZ

GregVK3kv, 2025-04-26:
I will have to load and test will try to complete tomorrow cheers GregVK3KV
Cheers,

Greg VK3KV

GregVK3kv, 2025-05-01:
I am away for 2 weeks OS will review on return.

The Scope display on transmit with PS implemented shows slight signals edge
side of bandpass. When PS off cleans up so maybe just a setting. Strange
why it's showing on display scope band skirts cleaner without PS?

Thetis same computer configuration extremely clean sharp band skirts.

Cheers thank you for all your great work I will be back on air testing 14th
May
Cheers,

Greg VK3KV

dl1bz, 2025-06-23:
Ok, no movement here. So I will close this issue now, because it is not really a bug or issue.
We can continue the discussion about it, for this please re-open a new discussion in the discussion tab subsection **Gerenal**.

73 Heiko, DL1BZ

closed 2025-06-23 by dl1bz