dl1bz/deskhpsdr #43: Unable to get sound working to Linux mint

genmce, 2025-07-04:
I have deskHPSDR compiled and working (with NR4 listed) on Lenovo t470.
I have headset pluged into 3.5mm jack 
I have been msg you on FB ask ki cha.
Linux Mint 22.1
Sound working via start up sound and test audio from sound settings.
I also confirm sound is working for other software, firefox browsing youtube.

inxi -A  =
API: AlSA
Server-1: PipeWire v:1.0.5 status:Active
Server-2: PulseAudio v:16.1 status:Active

I used pulseaudio --start to start the second.
I don't know how to select which server or is that what happens from  "AUDIO=" in make.config ?

deskHPSDR is running on WWV 10mhz - I see signal on the waterfall, S8 (-73 dBm)
ADC overdrive is red text though I can't quite read it due to it being cut in half. I adjust agc lower 
I have edited make.config.deskhpsdr 
attempting builds with 
AUDIO=PULSE
No audio output from deskHPSDR

AUDIO=PIPE   [should this be PIPEWIRE?]
No audio output from deskHPSDR
next to local mic shows built-in audio analog stereo

AUDIO=ALSA
No audio output from deskHPSDR

Suggestions?
Thank you

dl1bz, 2025-07-07:
No, Audio is only supported with three options:
PULSE (Pulseaudio, default or recommed if using Linux)
ALSA (as option to Pulseaudio if this cannot used by different reasons)
PORTAUDIO (Apples own audio layer if using macOS, not available if using Linux).

PipeWire, default installed with Linux Environments like Debian 12, has a wrapper to Pulseaudio. So use AUDIO=PULSE if want using Pipewire.

I'm running here Mint too (Debian version Faye) and Audio works great with Pulseaudio and Pipewire. With PiOS and my Pi5 I use Pulseaudio and Pipewire too, also without any issues.

First check with the shell, if there any audio devices present:
```
aplay -l
arecord -l
```

You must see your system audio devices.
**Please check too in deskHPSDR in RX Menu if Local Audio Output is ticked ON and you select your audio output device what you want to use.**

You can also install pasystray for control the pulseaudio things:
`sudo apt-get install pasystray`

You can also add two packages:
```
sudo apt-get install pavumeter
sudo apt-get install pavucontrol
```

So you can check, if there any audio devices work for playback and record. All audio devices need support 48kHz as sampling rate (depends from your audio hardware). Do you have an Headset with USB interface ? Try this too. I use here a Plantronics/Poly Headset Blackwire 3210 Mono with build-in USB audio interface, works great with my Linux-Systems (Notebooks with Mint and my Pi5).

73 Heiko, DL1BZ

genmce, 2025-07-08:
Thank you, this is working fine.
Have a great day!
73 de k5kip

closed 2025-07-08 by genmce